sq8vps/vp-digi #9: po polsku

sp00ck, 2022-09-24:
Poprosze o opis po polsku. 
nie zamiast ale jednak jako informacje. 

po drugie pytanie czy przejscie na teensy nie zwiekszyło by szybkości nadawania? tam sa dużo wieksze czestotliwości CPU

sq8vps, 2022-09-25:
Nie wiem, w jakim rozumieniu ma się zwiększyć szybkość nadawania. Jeśli chodzi o przejście na 9600 Bd to nie ma z tym problemu, bo procesor sobie spokojnie poradzi - zresztą mam napisany i sprawdzony modem 9600. Jednym z podstawowych założeń projektu była minimalna cena, więc wg mnie inne platformy odpadają.
Opis przetłumaczę na polski w wolnej chwili.

sp00ck, 2022-09-27:
Rozumiem, będę czekał na polski opis.

Czy można nadawac tym na wolnych pasmach? 466 np. ideałem było by użycie meshstatic 
Czy rozważałeś dodanie ekranu i 10 przycisków jak w starych komórkach słownik T9

SQ9MDD, 2022-10-01:
To jednak jest projekt modemu TNC z funkcjonalnościami DIGI. Protokół APRS enkapsulowany w protokole AX.25 kompletnie nie ma to nic wspólnego z meshtastic.

sq8vps, 2022-11-05:
Dla uzupełnienia - dodawanie ekranu i klawiatury zdecydowanie wykracza poza zakres tego projektu, bo powiększyło by go to chyba dwukrotnie - również w kwestii zajętości pamięci, która już jest praktycznie cała zajęta.
Miałem propozycje, żeby zrobić z tego tracker - i jest to na mojej liście.

closed 2022-11-05 by sq8vps

sp00ck, 2022-11-07:
TNC nie wyklucza meshstastic ;) chodzi o to by dało się wysyłać dowolne ramki.

Pamięć w 21 wieku jest problemem? Wystarczy przejść na rpi 2040 albo Ox64 albo PIC

Ja mogę tylko wypowiadać się jako przyszły użytkownik. Noszenie plątaniny kabli, jakiegoś dodatkowego urzadzenia itp. ( i tak trzeba nosić samo radio) to jednak zbyt dużo rzeczy, które może się zepsuc lub bedzie za skomplikowane do obsługi.

Zatem jak najbardziej otwarte, tak by każdy mógł w przyszłości użyć tego jak chce i taki protokół jak potrafi. 

P.S. Przydało by się też by to działało już, bo wiecie wojne mamy za miedzą