sq8vps/vp-digi #18: Add precise measured current (mA) values to readme

DeflateAwning, 2023-08-14:
Can you measure the min and max current draw of this device? We're considering using it for a cubesat project, where we have very tight power draw requirements

sq8vps, 2023-08-16:
The Blue Pill has no current measurement circuit built-in, but it is perfectly possible to include a shunt resistor in series with power supply and measure voltage drop with some op-amp. Currently I'm rewriting the code and voltage measurement and embedding it in beacon is on my to-do list and will be added in next release.

DeflateAwning, 2023-09-12:
The microcontroller uses about 40-50 mA in our testing, if that's a piece of information you want to add to the documentation

sq8vps, 2025-08-10:
Added, thank you.

closed 2025-08-10 by sq8vps