sq8vps/vp-digi #35: Missing rs.h - vp-digi failed to build

t52ta6ek, 2023-10-15:
Greetings!

Using STM32 IDE 1.13.2, Windows 10.
Cloned repository and imported into IDE
When attempting to reproduct the build, fx25.c repots failed to compile because of missing header file rs.h .

arm-none-eabi-gcc "../Core/Src/stm32f1xx_hal_msp.c" -mcpu=cortex-m3 -std=gnu11 -g -DDEBUG -DENABLE_FX25 -DUSE_HAL_DRIVER -DSTM32F103xB -c -I../Core/Inc -I../Drivers/STM32F1xx_HAL_Driver/Inc -I../Drivers/STM32F1xx_HAL_Driver/Inc/Legacy -I../Drivers/CMSIS/Device/ST/STM32F1xx/Include -I../Drivers/CMSIS/Include -I../lwfec -I../USB_DEVICE/App -I../USB_DEVICE/Target -I../Middlewares/ST/STM32_USB_Device_Library/Core/Inc -I../Middlewares/ST/STM32_USB_Device_Library/Class/CDC/Inc -Og -ffunction-sections -fdata-sections -Wall -fstack-usage -fcyclomatic-complexity -MMD -MP -MF"Core/Src/stm32f1xx_hal_msp.d" -MT"Core/Src/stm32f1xx_hal_msp.o" --specs=nano.specs -mfloat-abi=soft -mthumb -o "Core/Src/stm32f1xx_hal_msp.o"
../Core/Src/fx25.c:24:10: fatal error: rs.h: No such file or directory
   24 | #include "rs.h"
      |          ^~~~~~
compilation terminated.

sq8vps, 2023-10-16:
Did you execute
```
git submodule init
git submodule update
```
before compilation, as specified in readme?

t52ta6ek, 2023-10-16:
Mea culpa! RTFM.
Build successful after completing git submodule steps... ;-)
tnx!

closed 2023-10-16 by t52ta6ek