sq8vps/vp-digi #44: AIOC build - USB serial device cannot start

tristanmoney, 2025-06-15:
Hi Piotr,

I am trying to test the AIOC with VP-Digi and I was able to build and flash but get the following error with the serial port:

[image]

Any idea why? I assume this USB serial port can to be used to configure VP-Digi or use it as a KISS TNC, just like like blue pill.

Thanks,

Tristan

tristanmoney, 2025-08-12:
I have version 1.11.5 of the STM32CubeF3 drivers and USB device library, if that matters. I think I tried an older version back in June but wasn't able to get it working.

sq8vps, 2026-11-01:
We discussed it privately with Tristan. There are (and always have been) some weird problems with the USB stack library, especially in debug builds, that I wasn't able to solve on my side. In release builds this works mostly as expected, except for the problem occuring sometimes after saving the configuration, requiring the device to be physically reconnected. This is marked as a known bug.

closed 2026-11-01 by sq8vps