sq8vps/vp-digi #45: ax25.c broken for FX25

hiperiondev, 2025-07-06:
When enable FX25 in line 532:
	if(rx->rx != RX_STAGE_FX25_FRAME)
	{

this if have not close brace in the rest of code.

Gethub-prog, 2025-07-12:
Aprs using Baofeng 5RH pro can't be digitized on vpdigi, raw packets come in, but can't be digitized... I don't know if this is from the Baofeng firmware or from vpdigi

sq8vps, 2025-08-10:
Hi,
I'm pretty sure there is a closing bracket in line 614, which corresponds to the opening bracket in line 533 (FX.25) and line 599 (non-FX.25). Does this cause errors during compilation?

sq8vps, 2026-11-01:
There were no further comments on this, and I'm unable (nor I have ever been) to reproduce this issue with a fresh repo clone, so I'm closing the issue.

closed 2026-11-01 by sq8vps